definelicht/hlslib #2: passing a struct containing a POD array as an argument to HLS entry function

cwsmith, 2019-02-08:
The DataPack class from hlslib appears to support arrays whose size is defined at compile time.  Is there a way to pass a struct containing a pointer to a dynamically sized POD array to an HLS kernel?  

We wrote up a hello world code that passes a struct as a `void*` then casts it to a struct inside the kernel function:

https://github.com/cwsmith/hls_sandbox/blob/919c2113935b44b59989ff7c3d01cca2c678738a/hello.cc#L35

This will compile and run in emulation mode (see the `README.md`).  I suspect that accessing the pointer inside the `ints` struct is invalid inside the kernel without it having a port assigned to it.  Is that correct?

definelicht, 2019-02-09:
You should think of `DataPack` as a wrapper for SIMD operations, and a way to force HLS ports to have a certain size (hence the size must be known at compile-time).

Instead of using pointers, I recommend using indices, and then pass the array your indexing into directly to the kernel. Generally, any access to global memory (which is the only memory where pointers make sense) should be to an interface defined at compile time. So, instead of passing a `void*`, pass a pointer to your POD type. You might have to use the `DATA_PACK` pragma for this to work.

definelicht, 2019-02-21:
Closing this for now.

closed 2019-02-21 by definelicht