definelicht/hlslib #8: Using hlslib cmake script with SDx2017.4

salehjg, 2019-08-03:
I am trying to migrate my SDAccel project from GUI to CMake buildsystem. 
While trying to figure out how to use your CMake script, I noticed something weird was happening.

I have installed SDx2017.4 and when I run `cmake ..`, in function `sdaccel_target`, the if-statement that detects SDAccel versions, evaluates false so as a result, CMake script does not execute required commands to generate *.xclbin files needed for emulation or hardware targets.  

```if(((${SDAccel_MAJOR_VERSION} LESS 2018) AND (${SDAccel_MINOR_VERSION} LESS 3)) OR ${SDAccel_MAJOR_VERSION} LESS 2017)```

The problem is with:
```(${SDAccel_MINOR_VERSION} LESS 3)``` 
Obviously, for 2017.4, the minor version is 4.

With all the changes in the recent releases of SDAccel, I am not sure if this is intentional and the CMake script does not support 2017.4 or this is some sort of bug.

I appreciate your quick response on this.
Thank.

closed 2019-08-03 by salehjg

definelicht, 2019-08-03:
Hi @salehjg, 2017.4 was the version where the build system changed from a single phase to a two phase (compile + link) system. The script will still generate the commands necessary to build the xclbin, but it's a single command (e.g., `build_foo_hardware`) instead of two commands (`compile_foo_hardware` and `link_foo_hardware`). Although, I see that the commands are named a bit weirdly -- I will update this to match the Xilinx nomenclature.

salehjg, 2019-08-03:
Hi,
Thank you so much for your quick response even for a closed issue. I appreciate it.
As you mentioned I was wrong. It does run the right commands to generate xclbin file. spcl/gmem_hls project helped me a lot to understand it.