definelicht/hlslib #11: Compatibily of full() empty() VivadoHLS methods for Stream class

diamantopoulos, 2019-11-07:
It would be nice to add the full() and empty()methods, so that we have less changes in the code when we use hlslib to wrap hls streams (i.e. instead of having to change to Isfull() and isEmpty()). This addition to Stream.h is working for me:

```
  // Compatibility with Vivado HLS interface
  bool empty() {
    #pragma HLS INLINE
    return IsEmpty();
  }

  bool full() {
    #pragma HLS INLINE
    return IsFull();
  }
```
thanks
--d

definelicht, 2019-11-07:
@diamantopoulos Thanks for the suggestion, I added this with 8284bf2.

closed 2019-11-07 by definelicht

diamantopoulos, 2019-11-07:
@definelicht thx, can we also add the `read(var)` method. i.e. read with argument, in addition to `read()` already provided. This works for me:

```
  void read(T &output) {
    #pragma HLS INLINE
    output = read();
  }
```

definelicht, 2019-11-12:
@diamantopoulos Added this also

diamantopoulos, 2019-11-12:
@definelicht cool, thanks!

commit 8284bf2b (2019-11-07): Add empty() and full() compatibility methods [Fixes #11]