definelicht/hlslib #12: Error in IsEmpty() when HLSLIB_SYNTHESIS is set

mksit, 2019-11-17:
Hi, I got the following error when calling IsEmpty() method after the HLSLIB_SYNTHESIS is set.

```
../../include/hlslib/xilinx/Stream.h:453:12: error: 'this' argument to member function 'empty' has type 'const hls::stream<int>', but function is not marked
      const
    return stream_.empty();
```

The reason of this error is that a non-const method empty() is called from a const method IsEmpty(). I would like to know if the const qualifier is really necessary for this method.

definelicht, 2019-11-18:
Hi @mksit. I cannot reproduce this; at least on 2018.3 the `empty()`-method of `hls::stream` is marked as const, so `IsEmpty()` synthesizes as expected. What is your exact setup, and can you provide code to reproduce this?

mksit, 2019-11-19:
I tested it with Vivado HLS 2019.1. A minimal program that just invokes IsEmpty() fails to compile when HLSLIB_SYNTHESIS is set.

SimpleTest.cpp
```
#include "hlslib/xilinx/Stream.h"

int main() {
    hlslib::Stream<int, 10> strm;
    bool empty = strm.IsEmpty();
}
```

```
In file included from ../../../../SimpleTest.cpp:1:0:
/home/Workspace/hlslib/include/hlslib/xilinx/Stream.h: In instantiation of ‘bool hlslib::Stream<T, capacity, <anonymous> >::IsEmpty() const [with T = int; unsigned int capacityDefault = 10u; hlslib::Storage storage = (hlslib::Storage)0]’:
../../../../SimpleTest.cpp:12:26:   required from here
/home/Workspace/hlslib/include/hlslib/xilinx/Stream.h:459:26: error: passing ‘const hls::stream<int>’ as ‘this’ argument discards qualifiers [-fpermissive]
     return stream_.empty();
                          ^
In file included from /home/Workspace/hlslib/include/hlslib/xilinx/Stream.h:10:0,
                 from ../../../../SimpleTest.cpp:1:
/opt/Xilinx/Vivado/2019.1/include/hls_stream.h:175:10: note:   in call to ‘bool hls::stream<__STREAM_T__>::empty() [with __STREAM_T__ = int]’
     bool empty() {
          ^~~~~
make: *** [obj/SimpleTest.o] Error 1

```

definelicht, 2019-11-19:
This is not a bug: `HLSLIB_SYNTHESIS` should only be set when you're running synthesis. If you want to do software simulation, you should not define this macro.

When running synthesis, Vivado HLS does not use the `hls_stream.h` header, and you will thus not get this error.

You can have a look at `xilinx_test/CMakeLists.txt` for examples on how you can build, simulate, and synthesize codes with `hlslib`.

closed 2019-11-19 by definelicht

mksit, 2019-11-21:
The synthesis works when I set HLSLIB_SYNTHESIS, but I ran into the same error when I tried to do C/RTL cosimulation. I cannot find how to set the flags properly for cosimulation in your example.

I attach my simple testing code along with a tcl script. Could you please suggest how I should change the tcl script in order to get the cosimulation working?

[hlslib_test.zip](https://github.com/definelicht/hlslib/files/3876474/hlslib_test.zip)

definelicht, 2019-11-21:
Unfortunately Vivado HLS' cosimulation is not support by hlslib. This is because the internal version of clang used for cosimulation used to be behind the one used for synthesis, so it did not support the same language features. I have not looked into whether this has been updated in the meantime.

Instead, I advise to use SDAccel/Vitis' hardware emulation feature. This uses the OpenCL flow by executing emulation kernels, and can give you cycle accurate emulation, even emitting waveforms you can inspect.

Thus, the three stages of validation are:
1. Software simulation (i.e., just compile and run the code using `HLSLIB_DATAFLOW`-macros).
2. Hardware emulation using SDAccel/Vitis
3. Executing in hardware