definelicht/hlslib #20: Cmake build is broken for Intel FPGA

IuriiKobein, 2020-04-22:
Setup:

Ubuntu 18.04
Intel OpenCL SDK 18.1
cmake version 3.17

Scenario:

cmake  -Bbuild -GNinja .         
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ - works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Could NOT find Vitis (missing: Vitis_COMPILER Vitis_HLS Vitis_INCLUDE_DIRS Vitis_LIBRARIES Vitis_FLOATING_POINT_LIBRARY Vitis_VERSION Vitis_MAJOR_VERSION Vitis_MINOR_VERSION) 
-- Found IntelFPGAOpenCL: /home/rnd/intelFPGA/18.1/hld/bin/aocl  
-- Configuring done
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
stdc++_PATH (ADVANCED)
    linked by target "RunJacobi2D.exe" in directory /home/rnd/dev/third_parties/hlslib/intel_test

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.

definelicht, 2020-06-04:
Hi @IuriiKobein, sorry I haven't seen this, my notification settings were screwed up.

I cannot reproduce this, it looks like something that is linked into `RunJacobi2D.exe`.

Could you try to print what libraries CMake is linking by adding this to `intel_test/CMakeLists.txt` between line 39 and 40:
```cmake
message("${IntelFPGAOpenCL_LIBRARIES} ${CMAKE_THREAD_LIBS_INIT}")
```

The error message suggests something is being linked without having been found, perhaps related to pthreads since it's stdc++ (although it claims to have found it).

definelicht, 2020-07-28:
Closing due to inactivity and failure to reproduce. Will reopen if you can provide more information.

closed 2020-07-28 by definelicht