definelicht/hlslib #21: cosim_design with Vivado_hls 2019.2

diamantopoulos, 2020-05-30:
Hi, I was trying to use the library with Vivado HLS 2019.2 and it's surprising that I probably cannot run `cosim_design` independent of `cynth_design` from my tcl scipt. 

So every time I am calling `cosim_design` it is forcing `csynth_design`, even if I had previously run succesfully `csynth_design`. So I have the obvious issue of how to set the `-DHLSLIB_SYNTHESIS` flag, since for `cosim_desing` it had to be undefined and for `csynth_design` it has to be defined. 

Wondering if you have face this issue already and there is a quick workaround. 
thx 
--d

definelicht, 2020-06-04:
Hi @diamantopoulos. Vivado HLS cosimulation is not supported by hlslib. This is because you cannot pass separate flags to the software and hardware compilation flows, as you have discovered yourself. 

You can still do hardware emulation with hlslib, but you will have to use the SDAccel flow.

diamantopoulos, 2020-06-09:
Thanks a lot, @definelicht for the response. It is great that the intention of hlslib is to bring the problems of rtl/co-sim early at C++/simulation level. But when I integrate the library as part of a bigger design, then co-sim is the last necessary step before going to bitstream. I suppose that it is only a matter of flags so that in `synthesis` part of `cosim` , the `-DHLSLIB_SYNTHESIS` is added and in the `simulation` part it is undefined, right? But this means that it should have somehow supported by the tool (e.g. pass cflags in csynth, cosim, which is not supported).

definelicht, 2020-06-09:
Now that I think about it, I think the problem was even more fundamental: both the software and hardware compilation actually need `-DHLSLIB_SYNTHESIS`, because they have to compile exactly the same code. However, this means that you cannot use the parallel processing elements anymore whenever there is data flowing in the other direction.

I also seem to remember that, when I tested it last, the co-simulation flow used _an older version of clang_ which did not support the same set of features as the HLS frontend, in particular C++11 features.

What happens if you set `-DHLSLIB_SYNTHESIS` and run co-simulation? Where does it fail?

diamantopoulos, 2020-06-09:
So if I **DON'T** set `-DHLSLIB_SYNTHESIS` on `cosim` I am getting this:

```
INFO: [COSIM 212-302] Starting C TB testing ... 
./cosim.tv.exe: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.22' not found (required by ./cosim.tv.exe)
ERROR: [COSIM 212-359] Aborting co-simulation: C TB simulation failed, nonzero return value '1'.
ERROR: [COSIM 212-320] C TB testing failed, stop generating test vectors. Please check C TB or re-run cosim.
ERROR: [COSIM 212-5] *** C/RTL co-simulation file generation failed. ***
ERROR: [COSIM 212-4] *** C/RTL co-simulation finished: FAIL ***
command 'ap_source' returned error code
    while executing
"source run_hls.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 [list source $arg] "

INFO: [Common 17-206] Exiting vivado_hls at Tue Jun  9 13:11:20 2020...
Makefile:101: recipe for target 'cosim' failed
make: *** [cosim] Error 1
```

It seems `Vitis' libstdc++` is missing `GLIBCXX_3.4.22` : 
```
strings /usr/lib/x86_64-linux-gnu/libstdc++.so.6 | grep GLIBCXX
GLIBCXX_3.4
GLIBCXX_3.4.1
GLIBCXX_3.4.2
GLIBCXX_3.4.3
GLIBCXX_3.4.4
GLIBCXX_3.4.5
GLIBCXX_3.4.6
GLIBCXX_3.4.7
GLIBCXX_3.4.8
GLIBCXX_3.4.9
GLIBCXX_3.4.10
GLIBCXX_3.4.11
GLIBCXX_3.4.12
GLIBCXX_3.4.13
GLIBCXX_3.4.14
GLIBCXX_3.4.15
GLIBCXX_3.4.16
GLIBCXX_3.4.17
GLIBCXX_3.4.18
GLIBCXX_3.4.19
GLIBCXX_3.4.20
GLIBCXX_3.4.21
```

However, I was wrong that I `cosim` forces `csynth` every time. It was a bug in my scripts forcing that. Now I have a working solution where I firstly run cysnth with `-DHLSLIB_SYNTHESIS`  set and then `cosim` again with this set. I think the only thing I've changed was `cosim_design -compiler gcc`

I'm still testing it but it seems everything works till now with Vivado 2019.2: `csim` without `-DHLSLIB_SYNTHESIS` and `csynth` and `cosim` with it. 
This is my tcl snippet:
```
if { $hlsSim} {
  set hlslib_flags "-std=c++11 "
}
if { $hlsSyn || $hlsCoSim}  {
  set hlslib_flags "-std=c++11 -DHLSLIB_SYNTHESIS"
}
```
The flag is added in the command `add_files  file.cpp -cflags "${hlslib_flags}` -csimflags "${hlslib_flags}"  for every file using `hlslib`.

definelicht, 2020-06-09:
That's good news, maybe they finally updated the version of clang used by the co-simulation flow! Let me know if everything works, then I can say that hlslib "officially" supports co-simulation also ;-)

closed 2020-07-07 by definelicht