definelicht/hlslib #22: Dataflow and Template Functions

salehjg, 2020-09-04:
Hi,
I have been trying to use hlslib to create a dataflow scheme using template functions and noticed that template functions are not supported.

It is important not to have any wrappers for manually instantiated template functions, so the final user can easily change the number of PEs(or blocks or whatever...) from the CMake script.

I was hoping for something like this to be possible:
```
For( i=0; i<unitCount;i++){
#pragma hls unroll
    HLSLIB_DATAFLOW_FUNCTION( 
             MyTemplate<myParameter>(.....));
}
```
Templating the whole cpp file using cmake would be too messy and I do not know a way to do the same thing using compile time c++ or preprocessors.

Any suggestions?

definelicht, 2020-09-04:
Hi Saleh,

This is a limitation of Vivado HLS, since it is performed on the LLVM IR, meaning the code has to be valid C++ so it can be parsed with the clang frontend. As you noticed, this means you cannot use runtime variables as template arguments, even if they _will_ eventually be compile-time fixed by the HLS passes.

You can can achieve what you want with pure HLS by using recursive templates, see an [example here](https://github.com/spcl/stencil_hls/blob/c532db10f7b2cbb1b265f50a8b48a3c15df56c26/include/Compute.h#L212-L226). However, this does not play nicely with the `HLSLIB_DATAFLOW_FUNCTION` macro either.

Honestly, the cleanest solution for you is probably to drop the template arguments and pass `i*myParameter` as a function argument, if possible. Once this runs through HLS it will end up compile-time fixed in your code, such that each processing element is specialized. You can see an example of this in our GEMM code: [callsite](https://github.com/spcl/gemm_hls/blob/eeccad84ce6c8f054a2f42bdd6e07ed45e277c79/kernel/Top.cpp#L110-L117) and the [specialized function](https://github.com/spcl/gemm_hls/blob/eeccad84ce6c8f054a2f42bdd6e07ed45e277c79/kernel/Compute.cpp#L11-L230). `locationN` ends up being specialized for each processing element.

Let me know if this solves your issue!

salehjg, 2020-09-04:
Works like a charm, Thank you!

closed 2020-09-04 by salehjg