definelicht/hlslib #24: Another Project Which Uses HLSLIB

salehjg, 2020-12-29:
Hi,

I hope you're having a nice day.
I felt obliged to share my project which uses hlslib, as it helped a lot and simplified complicated procedures immensely. I cannot imagine developing the same project without hlslib and cmake.
https://github.com/salehjg/DGCNN-on-FPGA
( The project aims to accelerate a point cloud classifier network (dgcnn) using an FPGA and its preprint soon will be ready. )

definelicht, 2021-01-02:
Hi Saleh,

I'm delighted to hear that hlslib is/has been useful to your work.
Let me know when the preprint has been uploaded, then I will add it along with a link to your repository to the list.

definelicht, 2021-02-19:
@salehjg Any news on this?

definelicht, 2021-03-29:
Closing due to inactivity. @salehjg feel free to reopen this once you have a link to the preprint or published article.

closed 2021-03-29 by definelicht