definelicht/hlslib #28: Compatibility with Vitis 2020.2?

KerstinKeller, 2021-03-25:
This project seems very interesting, since I am trying to get up to speed developing SW for FPGA.
However I am using Vitis 2020.2 and had some problems to get it find my Vitis Installation.
Has the `FindVitis.cmake` been tested for 2020.2? 

When trying to run `cmake .. -DVITIS_ROOT=path\to\xilinx\Vitis\2020.2`  I get the error:
```
-- Could NOT find Vitis (missing: Vitis_COMPILER Vitis_HLS Vitis_INCLUDE_DIRS Vitis_LIBRARIES Vitis_FLOATING_POINT_LIBRARY Vitis_VERSION Vitis_MAJOR_VERSION Vitis_MINOR_VERSION Vitis_PLATFORMINFO)
-- Could NOT find IntelFPGAOpenCL (missing: IntelFPGAOpenCL_AOCL IntelFPGAOpenCL_AOC IntelFPGAOpenCL_INCLUDE_DIRS IntelFPGAOpenCL_LIBRARIES IntelFPGAOpenCL_RPATH IntelFPGAOpenCL_VERSION IntelFPGAOpenCL_MAJOR_VERSION IntelFPGAOpenCL_MINOR_VERSION)
```

Also I am interested only in the Vitis part, so is the IntelFPGA part strictly necessary or can I disable it?

definelicht, 2021-03-25:
Hi Kerstin -- I'm running Vitis 2020.2 myself, so it's expected to work. As far as I can see, you are invoking CMake correctly, assuming the `VITIS_ROOT` path is what hlslib expects.

I can't help but notice that you are passing what looks like a Windows path -- are you running this on Windows? I've never tested hlslib on Windows, and I wouldn't be surprised if there are some issues there.

As a sanity check, could you also verify that `path\to\xilinx\Vitis\2020.2\bin\v++` exists?

cothan, 2021-04-30:
I have similar problem as well, I'm running Vitis 2020.2. 
Here is my log:

```bash
cothan@vitis:~/work/hlslib/build$ cmake .. -DVITIS_ROOT=/vitis/install/V0.2//2020
-- Using user defined Vitis directory: /vitis/install/Vitis/2020.2/
CMake Warning at cmake/FindVitis.cmake:106 (message):
  vitis_hls used from Vitis 2020.1 and onwards introduces breaking changes to
  hls::stream.  Please pass -D__VITIS_HLS__ or -D__VIVADO_HLS__ in your
  synthesis script depending on which tool you are using to always use a
  working version.
Call Stack (most recent call first):
  CMakeLists.txt:17 (find_package)

CMake Error at cmake/FindVitis.cmake:198 (message):
  The Xilinx Runtime (XRT) was not found.  You can specify the XRT directory
  with the XRT_ROOT_DIR variable.
Call Stack (most recent call first):
  CMakeLists.txt:17 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/cothan/work/hlslib/build/CMakeFiles/CMakeOutput.log".
See also "/home/cothan/work/hlslib/build/CMakeFiles/CMakeError.log".
cothan@vitis:~/work/hlslib/build$ 
```

Any idea? 

Great project, btw.

definelicht, 2021-04-30:
Hi @cothan, this looks like a different issue. Did you try explicitly passing the path to an XRT installation as suggested in the error message?

You also shouldn't be getting that warning when using Vitis 2020.2. Could you do me a favor and check what `Vitis_VERSION`, `Vitis_MAJOR_VERSION` and `Vitis_MINOR_VERSION` are set to after running your CMake script? You can print them by adding this after line 87:

```cmake
message(STATUS "Version: ${Vitis_VERSION}, major: ${Vitis_MAJOR_VERSION}, minor ${Vitis_MINOR_VERSION}")
```

cothan, 2021-04-30:
Hi @definelicht , 

It's seem like the script is not able to find the Vitis version.. 

As shown in the picture, I set two directory paths, and `cd` to the real path to make sure it's exists. 

[image]

definelicht, 2021-05-01:
The thing is, even if the folder is somehow incorrect, it should be able to deduce the version number from the folder name, so I'm quite confused as to what is happening here. Unfortunately I can't reproduce it, so not sure how to debug. 

Are you running this on Linux, or some Windows setup?

Have you tried setting your `PATH` to point to the folder that `v++` is in _instead_ of setting `VITIS_ROOT`, i.e.:

```bash
PATH=/vitis/install/Vitis/2020.2/bin cmake ..
```

Did you try setting `XILINX_XRT` to point to the installation of XRT, like the error suggests? (This is different from your Vitis installation!)

closed 2021-06-17 by definelicht

madhureddyr, 2022-02-05:
C:\Users\madhu\Downloads\gemm_hls-master\gemm_hls-master\build>cmake ../ -DMM_DATA_TYPE=float -DMM_PARALLELISM_N=32 -DMM_PARALLELISM_M=8 -DMM_MEMORY_TILE_SIZE_N=512 -DMM_MEMORY_TILE_SIZE_M=512
-- Selecting Windows SDK version 10.0.19041.0 to target Windows 10.0.19043.
CMake Error at CMakeLists.txt:72 (find_package):
  Syntax error in cmake code at

    C:/Users/madhu/Downloads/gemm_hls-master/gemm_hls-master/CMakeLists.txt:72

  when parsing string

    C:\Xilinx\Vitis_HLS\2020.2

  Invalid escape sequence \X

-- Configuring incomplete, errors occurred!
See also "C:/Users/madhu/Downloads/gemm_hls-master/gemm_hls-master/build/CMakeFiles/CMakeOutput.log".

madhureddyr, 2022-02-05:
any please help me , i am new to make and Vivado.

madhureddyr, 2022-02-05:
> C:\Users\madhu\Downloads\gemm_hls-master\gemm_hls-master\build>cmake ../ -DMM_DATA_TYPE=float -DMM_PARALLELISM_N=32 -DMM_PARALLELISM_M=8 -DMM_MEMORY_TILE_SIZE_N=512 -DMM_MEMORY_TILE_SIZE_M=512 -- [...]

**i am using this in windows 10 with vivado 2020.2**

definelicht, 2022-02-05:
@madhureddyr I think the issue is backslashes in Windows file paths: https://stackoverflow.com/questions/13737370/cmake-error-invalid-escape-sequence-u/28565713

Can you see where in `FindVitis.cmake` this error is occurring?

madhureddyr, 2022-02-05:
I changed the path from backslashes to forward slashes,I am still getting the log as:
C:\Users\madhu\Downloads\gemm_hls-master\gemm_hls-master\build>cmake ../ -DMM_DATA_TYPE=float -DMM_PARALLELISM_N=32 -DMM_PARALLELISM_M=8 -DMM_MEMORY_TILE_SIZE_N=512 -DMM_MEMORY_TILE_SIZE_M=512
-- Selecting Windows SDK version 10.0.19041.0 to target Windows 10.0.19043.
CMake Warning at CMakeLists.txt:72 (find_package):
  By not providing "FindC:/Xilinx/Vitis_HLS/2020.2/bin.cmake" in
  CMAKE_MODULE_PATH this project has asked CMake to find a package
  configuration file provided by "C:/Xilinx/Vitis_HLS/2020.2/bin", but CMake
  did not find one.

  Could not find a package configuration file provided by
  "C:/Xilinx/Vitis_HLS/2020.2/bin" with any of the following names:

    C:/Xilinx/Vitis_HLS/2020.2/binConfig.cmake
    c:/xilinx/vitis_hls/2020.2/bin-config.cmake

  Add the installation prefix of "C:/Xilinx/Vitis_HLS/2020.2/bin" to
  CMAKE_PREFIX_PATH or set "C:/Xilinx/Vitis_HLS/2020.2/bin_DIR" to a
  directory containing one of the above files.  If
  "C:/Xilinx/Vitis_HLS/2020.2/bin" provides a separate development package or
  SDK, be sure it has been installed.

-- Looking for pthread.h
-- Looking for pthread.h - not found
-- Found Threads: TRUE
-- A library with BLAS API not found. Please specify library location.
CMake Error at CMakeLists.txt:105 (if):
  if given arguments:

    "(" "(" "LESS" "2018" ")" "AND" "(" "LESS" "3" ")" ")" "OR" "LESS" "2017"

  Unknown arguments specified

-- Configuring incomplete, errors occurred!
See also "C:/Users/madhu/Downloads/gemm_hls-master/gemm_hls-master/build/CMakeFiles/CMakeOutput.log".
See also "C:/Users/madhu/Downloads/gemm_hls-master/gemm_hls-master/build/CMakeFiles/CMakeError.log".

madhureddyr, 2022-02-05:
@definelicht 
please help with the issue.
Many thanks for considering my request.
Shall we connect via any medium to clear my problem...

definelicht, 2022-02-06:
Did you make changes to the CMake files in hlslib? The error message you got looks like you are changing the `find_package` macro.

madhureddyr, 2022-02-08:
there are no files in hslib folder.

madhureddyr, 2022-02-08:
> Did you make changes to the CMake files in hlslib? The error message you got looks like you are changing the `find_package` macro.

no, I did make changes in hslib folder.