definelicht/hlslib #38: Support for aarch64/arm64?

derpda, 2021-06-17:
Thanks for merging my PR so quickly!

I just ran into the next issue :smile: 

#### ARM64 support
As we are moving along with the project, we are trying to run embedded designs.
For that, we would need host code compiled for ARM64, but it seems like hlslib does not support that yet:
https://github.com/definelicht/hlslib/blob/e6b4c1d0ac01e91fe252ba4ee70407d8576bb3bc/cmake/FindVitis.cmake#L133-L137

What would need to be done to support ARM64?
It looks like we would need the right runtime and possibly the floating point libraries, but I really don't fully understand this part.

definelicht, 2021-06-17:
I would be happy to add ARM support, but since we don't use or have embedded boards in our lab, it's hard for me to test :-)

If you want to take a crack at it you're more than welcome, but then you might also be volunteering to maintain it ;-)

derpda, 2021-06-17:
I might just give it a shot...
I think I can figure out the xrt stuff, but I have never used sdaccel. 
Also, what is the floating point library thing for?

definelicht, 2021-06-17:
> Also, what is the floating point library thing for?

Half precision :-)

derpda, 2021-06-17:
Ohh that makes sense... in my defense, it was 4am over her!! :D

definelicht, 2021-06-18:
No criticism implied, just answered your question 😉

derpda, 2021-12-15:
Just came back to post a new issue, forgot this one was still open...
I think this is easy to work around (I've had no issues doing so), so I'll close this for now.

closed 2021-12-15 by derpda