definelicht/hlslib #43: Tight packing for DataPack on hardware

derpda, 2021-09-02:
In `DataPack`, the data is stored in an `ap_uint` of bit width `8*<bytes in T>`. 
https://github.com/definelicht/hlslib/blob/54b23af4c23dd6d6415c3a41567781f7eb8c7a56/include/hlslib/xilinx/DataPack.h#L25-L34
This of course is necessary on a CPU since it can only access memory on a resolution of bytes.

However, on an FPGA this introduces significant storage inefficiency.  I want to store an array of fixed-point data in the URAM of the FPGA. I have some liberty to decide the bit-width of a single data element. Considering that it is best to use power-of-two shaped accesses, we want to choose a bit width so that some power-of-two elements fit into 72 bits. This means we can choose 9, 18, 36 or 72 as the bit-width for best efficiency in the URAM. In our case, we were able to make do with 9bits.

My first try was thus doing something like the following:
```c++
hlslib::DataPack<ap_fixed<9, 1>, 8> array[N];
#pragma HLS bind_storage variable=array type=RAM_1P impl=URAM
```
Due to line 31 in the code I cited above however, the DataPack actually internally uses a `uint<128>` to store the 8 fixed-point numbers, not `uint<72>`.
This of course means I need twice as much URAM for this type of array.

To solve this for my case, I made my own tiny class (taking the important parts from your DataPack):
```c++
template<int width>
class TightPack {
 public:
  static constexpr int kWidth = width;
  #if defined(HLSLIB_SYNTHESIS)
  static constexpr int kBits = weight_t::width;
  #else
  static constexpr int kBits = 8 * sizeof(weight_t);
  #endif
 private:
  ap_uint<width*kBits> data_;
  using Pack_t = ap_uint<kBits>;
 public:
  weight_t Get(int i) const {
    // Just like regular DataPack
  }

  void Set(int i, weight_t val) {
    // Just like regular DataPack
  }
};
```
The only difference (aside from a lot of missing functionality)  is the calculation of `kWidth`, and I made that depend on whether we are in synthesis or not.
It is also specialized for `ap_fixed`. 

Ideally, I would like to make a small template specialization of your DataPack class instead, as in
```c++
template<int width>
class DataPack<ap_fixed> {
  // Only change kWidth implementation
};
```
However, this would mean I would need to copy the entire `DataPack` class content in there, defeating the purpose of specialization. 
Maybe putting a helper class like the following in between might improve this:
```c++
template<typename T>
WidthCalculator {
 public:
  static constexpr int value = 8  * sizeof(T);
};
```
DataPack could use this class to calculate kBits in line 31 like:
```c++
  static constexpr int kBits = WidthCalculator<T>::value;
```

Then I could subclass the WidthCalculator to my liking, for example
```c++
template<>
template<int W, int I>
WidthCalculator<ap_fixed<W, I> {
 public:
  static constexpr int value = W;
};
```
and use `DataPack<ap_fixed<9, 1>, 8>` for optimal storage in URAM (and BRAM for that matter).

Actually, might this specialization (and similar ones for the other fixed width types in HLS) be important enough to include in the library itself?

definelicht, 2021-09-02:
I think we should go with the width calculator approach. I've commented on the PR.

definelicht, 2021-09-22:
Solved with #47

closed 2021-09-22 by definelicht