definelicht/hlslib #46: hlslib::ConstLog2 off by 1?

derpda, 2021-09-15:
ConstLog2 is defined as
https://github.com/definelicht/hlslib/blob/521363f075e554def750e2e4557819d7e0253325/include/hlslib/xilinx/Utility.h#L14-L16

For an input of (for example) 16 this yields 5, which is wrong I believe. It should be 4.

A correct definition might be:
```
constexpr unsigned char ConstLog2(unsigned long val) {
  return val < 2 ? 0 : 1 + ConstLog2(val >> 1);
}
```
This would also return 0 for an input of 0, and correctly return 0 for 1, and also correct values for all higher input values I think.

definelicht, 2021-09-15:
You are right! Do you want to make a PR, or should I just push a fix?

derpda, 2021-09-21:
ConstLog2 seems to be used in a few places that I don't fully understand (namely Flatten and ShiftRegister), and just changing the implementation causes the tests to fail. 
I could just add a `+1` to any place `ConstLog2` is used, but that seems like a hack.
Could you have a look?

definelicht, 2021-09-21:
Fixed in #49

closed 2021-09-21 by definelicht