definelicht/hlslib #51: c++0x_warning.h

sarahelghandour, 2021-11-05:
Hi,

I have watched the "Productive Parallel Programming for FPGA with High Level Synthesis" on youtube and I found it be very useful, thank you for posting it! I'm trying to run the examples on my machine but I have encountered the warning and error shown below when synthesizing any example that includes the use of hlslib (this is from Example 4). I'm using vivado hls 2019.2 on Windows. I have tried setting the CFLAG to -std=c++0x or -std=c++11, when doing so too many errors arise as shown in the second picture below. When I open the c++0x_warning.h file the warning is mentioning c++11. I've spend quite sometime trying to solve this issue to no avail. 

The C simulation runs successfully though. 

Thank you for your help.

Best Regards,
Sarah

"
Starting C synthesis ...
C:/Xilinx/Vivado/2019.2/bin/vivado_hls.bat C:/Users/Sara/AppData/Roaming/Xilinx/Vivado/Example4/Example4/csynth.tcl
INFO: [HLS 200-10] Running 'C:/Xilinx/Vivado/2019.2/bin/unwrapped/win64.o/vivado_hls.exe'
INFO: [HLS 200-10] For user 'Sara' on host 'lenovo-sara' (Windows NT_amd64 version 6.2) on Fri Nov 05 14:09:18 +1100 2021
INFO: [HLS 200-10] In directory 'C:/Users/Sara/AppData/Roaming/Xilinx/Vivado'
Sourcing Tcl script 'C:/Users/Sara/AppData/Roaming/Xilinx/Vivado/Example4/Example4/csynth.tcl'
INFO: [HLS 200-10] Opening project 'C:/Users/Sara/AppData/Roaming/Xilinx/Vivado/Example4'.
INFO: [HLS 200-10] Adding design file 'Example4/Example4/Example4.cpp' to the project
INFO: [HLS 200-10] Adding test bench file 'Example4/Example4/Test.cpp' to the project
INFO: [HLS 200-10] Opening solution 'C:/Users/Sara/AppData/Roaming/Xilinx/Vivado/Example4/Example4'.
INFO: [SYN 201-201] Setting up clock 'default' with a period of 10ns.
WARNING: [HLS 200-40] Cannot find library 'C:/Xilinx/Vivado/2019.2/common/technology/xilinx/Virtex-7/Virtex-7.lib'.
WARNING: [HLS 200-40] Cannot find library 'xilinx/Virtex-7/Virtex-7'.
INFO: [HLS 200-10] Setting target device to 'xc7vx485t-ffg1157-1'
INFO: [SYN 201-201] Setting up clock 'default' with a period of 10ns.
INFO: [SCHED 204-61] Option 'relax_ii_for_timing' is enabled, will increase II to preserve clock frequency constraints.
INFO: [HLS 200-10] Analyzing design file 'Example4/Example4/Example4.cpp' ... 
WARNING: [HLS 200-40] In file included from Example4/Example4/Example4.cpp:1:
In file included from Example4/Example4/Example4.h:3:
In file included from Example4/Example4/hlslib/xilinx/Stream.h:11:
In file included from C:/Xilinx/Vivado/2019.2/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\condition_variable:34:
C:/Xilinx/Vivado/2019.2/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\bits/c++0x_warning.h:31:2: error: This file requires compiler and library support for the upcoming ISO C++ standard, C++0x. This support is currently experimental, and must be enabled with the -std=c++0x or -std=gnu++0x compiler options.
#error This file requires compiler and library support for the upcoming \
 ^
1 error generated.
C preprocessor failed.
    while executing
"source C:/Users/Sara/AppData/Roaming/Xilinx/Vivado/Example4/Example4/csynth.tcl"
    invoked from within
"hls::main C:/Users/Sara/AppData/Roaming/Xilinx/Vivado/Example4/Example4/csynth.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel 1 hls::main {*}$args"
    (procedure "hls_proc" line 5)
    invoked from within
"hls_proc $argv"
Finished C synthesis.
"

[image]

[image]

definelicht, 2021-11-05:
Hi Sarah,

When you run synthesis, both the `std=c++11` and the `-DHLSLIB_SYNTHESIS` flags should be set. The error you are seeing is because the simulation environment uses C++ locks and mutexes in simulation mode, which will be removed when running for synthesis. This is only done when the `HLSLIB_SYNTHESIS` macro is set.

sarahelghandour, 2021-11-06:
Thank you very much, I've added the -DHLSLIB_SYNTHESIS flag as well as std=c++11 and that solved this issue. I'm getting now another error which after some research it seems that the HLS bind_storage pragma is not supported in 2019.2? Is my understanding correct or am I missing something? I'm new to Vivado HLS. The error message is shown below. 

If I'm going to move to a newer version, which version do you recommend that would fully support hlslib?

Thanks again for your help, much appreciated. 

Starting C synthesis ...
C:/Xilinx/Vivado/2019.2/bin/vivado_hls.bat C:/Users/Sara/AppData/Roaming/Xilinx/Vivado/Example4/Example4/csynth.tcl
INFO: [HLS 200-10] Running 'C:/Xilinx/Vivado/2019.2/bin/unwrapped/win64.o/vivado_hls.exe'
INFO: [HLS 200-10] For user 'Sara' on host 'lenovo-sara' (Windows NT_amd64 version 6.2) on Sat Nov 06 07:58:26 +1100 2021
INFO: [HLS 200-10] In directory 'C:/Users/Sara/AppData/Roaming/Xilinx/Vivado'
Sourcing Tcl script 'C:/Users/Sara/AppData/Roaming/Xilinx/Vivado/Example4/Example4/csynth.tcl'
INFO: [HLS 200-10] Opening project 'C:/Users/Sara/AppData/Roaming/Xilinx/Vivado/Example4'.
INFO: [HLS 200-10] Adding design file 'Example4/Example4/Example4.cpp' to the project
INFO: [HLS 200-10] Adding test bench file 'Example4/Example4/Test.cpp' to the project
INFO: [HLS 200-10] Opening solution 'C:/Users/Sara/AppData/Roaming/Xilinx/Vivado/Example4/Example4'.
INFO: [SYN 201-201] Setting up clock 'default' with a period of 10ns.
WARNING: [HLS 200-40] Cannot find library 'C:/Xilinx/Vivado/2019.2/common/technology/xilinx/Virtex-7/Virtex-7.lib'.
WARNING: [HLS 200-40] Cannot find library 'xilinx/Virtex-7/Virtex-7'.
INFO: [HLS 200-10] Setting target device to 'xc7vx485t-ffg1157-1'
INFO: [SCHED 204-61] Option 'relax_ii_for_timing' is enabled, will increase II to preserve clock frequency constraints.
INFO: [HLS 200-10] Analyzing design file 'Example4/Example4/Example4.cpp' ... 
WARNING: [HLS 214-114] Since the only kind of statements allowed in a canonical dataflow region are variable declarations and function calls, the compiler may not be able to correctly handle the region: Example4/Example4/Example4.cpp:60:25
WARNING: [HLS 214-114] Since the only kind of statements allowed in a canonical dataflow region are variable declarations and function calls, the compiler may not be able to correctly handle the region: Example4/Example4/Example4.cpp:67:3
WARNING: [HLS 214-114] Since the only kind of statements allowed in a canonical dataflow region are variable declarations and function calls, the compiler may not be able to correctly handle the region: Example4/Example4/Example4.cpp:73:29
WARNING: [HLS 200-471] Dataflow form checks found 3 issue(s) in file Example4/Example4/Example4.cpp
ERROR: [HLS 200-70] '#pragma HLS bind_storage variable=&stream_ type=FIFO impl=BRAM' is not a valid pragma.
ERROR: [HLS 200-70] '#pragma HLS bind_storage variable=&stream_ type=FIFO impl=LUTRAM' is not a valid pragma.
ERROR: [HLS 200-70] '#pragma HLS bind_storage variable= &stream_ type=FIFO impl=SRL' is not a valid pragma.
Pragma processor failed: 
    while executing
"source C:/Users/Sara/AppData/Roaming/Xilinx/Vivado/Example4/Example4/csynth.tcl"
    invoked from within
"hls::main C:/Users/Sara/AppData/Roaming/Xilinx/Vivado/Example4/Example4/csynth.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel 1 hls::main {*}$args"
    (procedure "hls_proc" line 5)
    invoked from within
"hls_proc $argv"
Finished C synthesis.

definelicht, 2021-11-06:
I tried compiling the examples with 2019.2, but I cannot reproduce this error. Did you modify the code in any way? Are you using the newest version of the examples repository?

I'm currently using Vitis 2021.1, but I would just recommend using the newest version available (currently 2021.2).

sarahelghandour, 2021-11-07:
No I didn't modify anything, and I got the code from https://github.com/spcl/hls_tutorial_examples just a few days ago so I think I have the latest version. I have tried synthesizing Example2 instead and still got the same errors. All I have done is place the hlslib files and Example2.cpp, Entry.cpp,Test.cpp (Example2 and Entry as source, Test as test bench) files  in the project folder and defined the top function to be Example2.

Starting C synthesis ...
C:/Xilinx/Vivado/2019.2/bin/vivado_hls.bat C:/Users/Sara/AppData/Roaming/Xilinx/Vivado/Example2/solution1/csynth.tcl
INFO: [HLS 200-10] Running 'C:/Xilinx/Vivado/2019.2/bin/unwrapped/win64.o/vivado_hls.exe'
INFO: [HLS 200-10] For user 'Sara' on host 'lenovo-sara' (Windows NT_amd64 version 6.2) on Sun Nov 07 22:01:16 +1100 2021
INFO: [HLS 200-10] In directory 'C:/Users/Sara/AppData/Roaming/Xilinx/Vivado'
Sourcing Tcl script 'C:/Users/Sara/AppData/Roaming/Xilinx/Vivado/Example2/solution1/csynth.tcl'
INFO: [HLS 200-10] Opening project 'C:/Users/Sara/AppData/Roaming/Xilinx/Vivado/Example2'.
INFO: [HLS 200-10] Adding design file 'Example2/Entry.cpp' to the project
INFO: [HLS 200-10] Adding design file 'Example2/Example2.cpp' to the project
INFO: [HLS 200-10] Adding test bench file 'Example2/Test.cpp' to the project
INFO: [HLS 200-10] Opening solution 'C:/Users/Sara/AppData/Roaming/Xilinx/Vivado/Example2/solution1'.
INFO: [SYN 201-201] Setting up clock 'default' with a period of 10ns.
WARNING: [HLS 200-40] Cannot find library 'C:/Xilinx/Vivado/2019.2/common/technology/xilinx/Virtex-7/Virtex-7.lib'.
WARNING: [HLS 200-40] Cannot find library 'xilinx/Virtex-7/Virtex-7'.
INFO: [HLS 200-10] Setting target device to 'xc7vx485t-ffg1157-1'
INFO: [SCHED 204-61] Option 'relax_ii_for_timing' is enabled, will increase II to preserve clock frequency constraints.
INFO: [HLS 200-10] Analyzing design file 'Example2/Example2.cpp' ... 
ERROR: [HLS 200-70] '#pragma HLS bind_storage variable=this->stream_ type=FIFO impl=BRAM' is not a valid pragma.
ERROR: [HLS 200-70] '#pragma HLS bind_storage variable=this->stream_ type=FIFO impl=LUTRAM' is not a valid pragma.
ERROR: [HLS 200-70] '#pragma HLS bind_storage variable=this->stream_ type=FIFO impl=SRL' is not a valid pragma.
Pragma processor failed: 
    while executing
"source C:/Users/Sara/AppData/Roaming/Xilinx/Vivado/Example2/solution1/csynth.tcl"
    invoked from within
"hls::main C:/Users/Sara/AppData/Roaming/Xilinx/Vivado/Example2/solution1/csynth.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel 1 hls::main {*}$args"
    (procedure "hls_proc" line 5)
    invoked from within
"hls_proc $argv"
Finished C synthesis.

definelicht, 2021-11-07:
I'm not sure why it compiles this part when you don't specify the resource, but the error is because the `bind_storage` was only introduced in a later version of Vitis. Is it possible for you to upgrade to the latest version?

sarahelghandour, 2021-11-09:
Hi,

Sorry for the late reply, I have moved to Vitis HLS 2021.1 and thankfully this error doesn't show anymore when I run C synthesis. Now, when I run the C simulation for example 2 it works fine as well, but for example 6 it generates the below message, and it seems like the simulation gets stuck in the case of Example 7. Is there a cflag I'm supposed to set for C simulation as well? The following flags are set be default " -cflags -Wno-unknown-pragmas -csimflags -Wno-unknown-pragmas ". Thanks a lot for your help.

Example6:
"Starting C simulation ...
C:/Xilinx/Vitis_HLS/2021.1/bin/vitis_hls.bat C:/Users/Sara/AppData/Roaming/Xilinx/Vitis/Example6/solution1/csim.tcl
INFO: [HLS 200-10] Running 'C:/Xilinx/Vitis_HLS/2021.1/bin/unwrapped/win64.o/vitis_hls.exe'
INFO: [HLS 200-10] For user 'Sara' on host 'lenovo-sara' (Windows NT_amd64 version 6.2) on Tue Nov 09 11:51:50 +1100 2021
INFO: [HLS 200-10] In directory 'C:/Users/Sara/AppData/Roaming/Xilinx/Vitis'
Sourcing Tcl script 'C:/Users/Sara/AppData/Roaming/Xilinx/Vitis/Example6/solution1/csim.tcl'
INFO: [HLS 200-1510] Running: open_project Example6 
INFO: [HLS 200-10] Opening project 'C:/Users/Sara/AppData/Roaming/Xilinx/Vitis/Example6'.
INFO: [HLS 200-1510] Running: set_top Example6_Vectorized 
INFO: [HLS 200-1510] Running: add_files Example6/Example6_Vectorized.cpp -cflags -DHLSLIB_SYNTHESIS -std=c++11 
INFO: [HLS 200-10] Adding design file 'Example6/Example6_Vectorized.cpp' to the project
INFO: [HLS 200-1510] Running: add_files Example6/Example6.cpp -cflags -DHLSLIB_SYNTHESIS -std=c++11 
INFO: [HLS 200-10] Adding design file 'Example6/Example6.cpp' to the project
INFO: [HLS 200-1510] Running: add_files Example6/Entry.cpp -cflags -DHLSLIB_SYNTHESIS -std=c++11 
INFO: [HLS 200-10] Adding design file 'Example6/Entry.cpp' to the project
INFO: [HLS 200-1510] Running: add_files -tb Example6/Test.cpp -cflags -Wno-unknown-pragmas -csimflags -Wno-unknown-pragmas 
INFO: [HLS 200-10] Adding test bench file 'Example6/Test.cpp' to the project
INFO: [HLS 200-1510] Running: open_solution solution1 -flow_target vivado 
INFO: [HLS 200-10] Opening solution 'C:/Users/Sara/AppData/Roaming/Xilinx/Vitis/Example6/solution1'.
INFO: [SYN 201-201] Setting up clock 'default' with a period of 10ns.
INFO: [HLS 200-1611] Setting target device to 'xcvu11p-flga2577-1-e'
INFO: [HLS 200-1505] Using flow_target 'vivado'
Resolution: For help on HLS 200-1505 see www.xilinx.com/cgi-bin/docs/rdoc?v=2021.1;t=hls+guidance;d=200-1505.html
INFO: [HLS 200-1510] Running: set_part xcvu11p-flga2577-1-e 
INFO: [HLS 200-1510] Running: create_clock -period 10 -name default 
INFO: [HLS 200-1510] Running: csim_design -quiet 
INFO: [SIM 211-2] *************** CSIM start ***************
INFO: [SIM 211-4] CSIM will launch GCC as the compiler.
   Compiling ../../../Test.cpp in debug mode
   Generating csim.exe
In file included from C:/Xilinx/Vitis_HLS/2021.1/include/floating_point_v7_0_bitacc_cmodel.h:144:0,
                 from C:/Xilinx/Vitis_HLS/2021.1/include/hls_fpo.h:189,
                 from C:/Xilinx/Vitis_HLS/2021.1/include/hls_half_fpo.h:64,
                 from C:/Xilinx/Vitis_HLS/2021.1/include/hls_half.h:71,
                 from C:/Xilinx/Vitis_HLS/2021.1/include/etc/ap_private.h:91,
                 from C:/Xilinx/Vitis_HLS/2021.1/include/ap_common.h:653,
                 from C:/Xilinx/Vitis_HLS/2021.1/include/ap_fixed.h:55,
                 from ../../../hlslib/xilinx/DataPack.h:8,
                 from ../../../Example6.h:3,
                 from ../../../Test.cpp:6:
C:/Xilinx/Vitis_HLS/2021.1/include/gmp.h:63:0: warning: "__GMP_LIBGMP_DLL" redefined
 #define __GMP_LIBGMP_DLL  0
 
In file included from C:/Xilinx/Vitis_HLS/2021.1/include/hls_fpo.h:189:0,
                 from C:/Xilinx/Vitis_HLS/2021.1/include/hls_half_fpo.h:64,
                 from C:/Xilinx/Vitis_HLS/2021.1/include/hls_half.h:71,
                 from C:/Xilinx/Vitis_HLS/2021.1/include/etc/ap_private.h:91,
                 from C:/Xilinx/Vitis_HLS/2021.1/include/ap_common.h:653,
                 from C:/Xilinx/Vitis_HLS/2021.1/include/ap_fixed.h:55,
                 from ../../../hlslib/xilinx/DataPack.h:8,
                 from ../../../Example6.h:3,
                 from ../../../Test.cpp:6:
C:/Xilinx/Vitis_HLS/2021.1/include/floating_point_v7_0_bitacc_cmodel.h:136:0: note: this is the location of the previous definition
 #define __GMP_LIBGMP_DLL 1
 
Test ran successfully.
INFO: [SIM 211-1] CSim done with 0 errors.
INFO: [SIM 211-3] *************** CSIM finish ***************
INFO: [HLS 200-111] Finished Command csim_design CPU user time: 1 seconds. CPU system time: 0 seconds. Elapsed time: 42.314 seconds; current allocated memory: 122.639 MB.
Finished C simulation.
"

sarahelghandour, 2021-11-09:
To give a complete picture, the below is the output of C synthesis of Example 7 which runs successfully, however, you can see many warnings like "WARNING: [HLS 200-805] An internal stream 'a_pipes.stream_.V.0' with default size can result in deadlock. Please consider resizing the stream using the directive 'set_directive_stream' or the 'HLS stream' pragma."
"
Starting C synthesis ...
C:/Xilinx/Vitis_HLS/2021.1/bin/vitis_hls.bat C:/Users/Sara/AppData/Roaming/Xilinx/Vitis/Example7/solution1/csynth.tcl
INFO: [HLS 200-10] Running 'C:/Xilinx/Vitis_HLS/2021.1/bin/unwrapped/win64.o/vitis_hls.exe'
INFO: [HLS 200-10] For user 'Sara' on host 'lenovo-sara' (Windows NT_amd64 version 6.2) on Tue Nov 09 12:23:10 +1100 2021
INFO: [HLS 200-10] In directory 'C:/Users/Sara/AppData/Roaming/Xilinx/Vitis'
Sourcing Tcl script 'C:/Users/Sara/AppData/Roaming/Xilinx/Vitis/Example7/solution1/csynth.tcl'
INFO: [HLS 200-1510] Running: open_project Example7 
INFO: [HLS 200-10] Opening project 'C:/Users/Sara/AppData/Roaming/Xilinx/Vitis/Example7'.
INFO: [HLS 200-1510] Running: set_top Example7 
INFO: [HLS 200-1510] Running: add_files Example7/Example7.cpp -cflags -DHLSLIB_SYNTHESIS -std=c++11 
INFO: [HLS 200-10] Adding design file 'Example7/Example7.cpp' to the project
INFO: [HLS 200-1510] Running: add_files -tb Example7/Test.cpp -cflags -Wno-unknown-pragmas -csimflags -Wno-unknown-pragmas 
INFO: [HLS 200-10] Adding test bench file 'Example7/Test.cpp' to the project
INFO: [HLS 200-1510] Running: open_solution solution1 -flow_target vivado 
INFO: [HLS 200-10] Opening solution 'C:/Users/Sara/AppData/Roaming/Xilinx/Vitis/Example7/solution1'.
INFO: [SYN 201-201] Setting up clock 'default' with a period of 10ns.
INFO: [HLS 200-1611] Setting target device to 'xcvu11p-flga2577-1-e'
INFO: [HLS 200-1505] Using flow_target 'vivado'
Resolution: For help on HLS 200-1505 see www.xilinx.com/cgi-bin/docs/rdoc?v=2021.1;t=hls+guidance;d=200-1505.html
INFO: [HLS 200-1510] Running: set_part xcvu11p-flga2577-1-e 
INFO: [HLS 200-1510] Running: create_clock -period 10 -name default 
INFO: [HLS 200-1510] Running: set_directive_top -name Example7 Example7 
INFO: [HLS 200-1510] Running: csynth_design 
INFO: [HLS 200-111] Finished File checks and directory preparation: CPU user time: 0 seconds. CPU system time: 0 seconds. Elapsed time: 0.112 seconds; current allocated memory: 1013.880 MB.
INFO: [HLS 200-10] Analyzing design file 'Example7/Example7.cpp' ... 
WARNING: [HLS 207-5528] Ignore interface attribute or pragma which is not used in top function: Example7/Example7.cpp:131:9
WARNING: [HLS 214-114] Since the only kind of statements allowed in a canonical dataflow region are variable declarations and function calls, the compiler may not be able to correctly handle the region: Example7/Example7.cpp:142:3
Resolution: For help on HLS 214-114 see www.xilinx.com/cgi-bin/docs/rdoc?v=2021.1;t=hls+guidance;d=214-114.html
WARNING: [HLS 200-471] Dataflow form checks found 1 issue(s) in file Example7/Example7.cpp
Resolution: For help on HLS 200-471 see www.xilinx.com/cgi-bin/docs/rdoc?v=2021.1;t=hls+guidance;d=200-471.html
WARNING: [HLS 207-5301] unused parameter 'print': C:/Xilinx/Vitis_HLS/2021.1/common/technology/autopilot\ap_int_base.h:792:16
INFO: [HLS 200-111] Finished Source Code Analysis and Preprocessing: CPU user time: 3 seconds. CPU system time: 1 seconds. Elapsed time: 14.06 seconds; current allocated memory: 124.610 MB.
INFO: [HLS 200-777] Using interface defaults for 'Vivado' flow target.
INFO: [HLS 214-279] Initial Interval estimation mode is set into default.
INFO: [HLS 214-284] Auto array partition mode is set into default.
INFO: [HLS 214-131] Inlining function 'hls::stream<float, 0>::stream()' into 'hlslib::Stream<float, 0ul, (hlslib::Storage)0>::Stream(char const*, unsigned long, hlslib::Storage)' (Example7/hlslib/xilinx/Stream.h:112:48)
INFO: [HLS 214-131] Inlining function 'hlslib::Stream<float, 0ul, (hlslib::Storage)0>::Stream(char const*, unsigned long, hlslib::Storage)' into 'hlslib::Stream<float, 0ul, (hlslib::Storage)0>::Stream(char const*)' (Example7/hlslib/xilinx/Stream.h:91:36)
INFO: [HLS 214-131] Inlining function 'hlslib::Stream<float, 0ul, (hlslib::Storage)0>::Stream(char const*)' into 'hlslib::Stream<float, 0ul, (hlslib::Storage)0>::Stream()' (Example7/hlslib/xilinx/Stream.h:83:14)
INFO: [HLS 214-131] Inlining function 'hls::stream<hlslib::DataPack<float, 4>, 0>::stream()' into 'hlslib::Stream<hlslib::DataPack<float, 4>, 0ul, (hlslib::Storage)0>::Stream(char const*, unsigned long, hlslib::Storage)' (Example7/hlslib/xilinx/Stream.h:112:48)
INFO: [HLS 214-131] Inlining function 'hlslib::Stream<hlslib::DataPack<float, 4>, 0ul, (hlslib::Storage)0>::Stream(char const*, unsigned long, hlslib::Storage)' into 'hlslib::Stream<hlslib::DataPack<float, 4>, 0ul, (hlslib::Storage)0>::Stream(char const*)' (Example7/hlslib/xilinx/Stream.h:91:36)
INFO: [HLS 214-131] Inlining function 'hlslib::Stream<hlslib::DataPack<float, 4>, 0ul, (hlslib::Storage)0>::Stream(char const*)' into 'hlslib::Stream<hlslib::DataPack<float, 4>, 0ul, (hlslib::Storage)0>::Stream()' (Example7/hlslib/xilinx/Stream.h:83:14)
INFO: [HLS 214-131] Inlining function 'hls::stream<float, 0>::write(float const&)' into 'hlslib::Stream<float, 0ul, (hlslib::Storage)0>::WriteBlocking(float const&, unsigned long)' (Example7/hlslib/xilinx/Stream.h:406:10)
[... cut ...]
INFO: [VLOG 209-307] Generating Verilog RTL for Example7.
INFO: [HLS 200-790] **** Loop Constraint Status: All loop constraints were satisfied.
INFO: [HLS 200-789] **** Estimated Fmax: 136.99 MHz
INFO: [HLS 200-111] Finished Command csynth_design CPU user time: 31 seconds. CPU system time: 8 seconds. Elapsed time: 63.454 seconds; current allocated memory: 415.399 MB.
INFO: [HLS 200-112] Total CPU user time: 33 seconds. Total CPU system time: 12 seconds. Total elapsed time: 67.536 seconds; peak allocated memory: 1013.880 MB.
Finished C synthesis.
"

definelicht, 2021-11-10:
As a fair warning, Vitis' own C-simulation does not always work particularly well, I favor compiling my programs with a regular C++ compiler when running simulation. This is all set up in the makefiles of the tutorial repository, but I realize that you are using the Xilinx GUI instead, which might not easily allow this.

Could it be that it's simply taking a very long time? These examples are more meant for showing the output of synthesis, rather than running them on large input sizes (which will be fast in hardware, but not in simulation).

Could you try setting the domain size to something relatively small (for example 64x64x64) in Example6.h or Example7.h?

sarahelghandour, 2021-11-11:
Thank you, changing the domain size didn't occur to me, I changed it to a smaller size and the C simulation ran successfully! (although the mentioned warning is still showing but I guess that's not a big issue). Thanks a lot for your help.

definelicht, 2021-11-11:
The warning is only relevant in hardware, and indeed it might benefit from making the FIFOs a bit deeper, but it should be safe for you to ignore :-)

closed 2021-11-11 by definelicht