definelicht/hlslib #59: `hlslib::Stream` with `ap_axiu`

derpda, 2022-01-26:
### Intro
Happy (belated) New Year!

I am still using hlslib frequently, so once again thanks for your work.

### Problem
Currently I am using AXI4-Stream interfaces between kernels, and that works as expected with the `hlslib::Stream` class.
However, I am now using an IP that requires a full AXI4-Stream interface, and thus I use `ap_axiu` to make HLS synthesize the required side channels.
For some reason the build fails when using `hlslib::Stream`, but works if I just replace those streams using `ap_axiu` with `hls::stream`.
There seem to be no useful messages in the logs as to what the cause is, it just states 
```
Failed to build kernel(ip) debug, see log for details: <path>/_x/debug_hw/debug/vitis_hls.log
```
Said log only tells me
```
INFO: [HLS 200-111] Finished Source Code Analysis and Preprocessing: CPU user time: 27.32 seconds. CPU system time: 1.27 seconds. Elapsed time: 27.56 seconds; current allocated memory: 206.791 MB.
INFO: [HLS 200-777] Using interface defaults for 'Vitis' flow target.
ERROR: [HLS 200-1715] Encountered problem during source synthesis
```
There are also no pertinent earlier warnings.

### Question
Would you happen to know what the cause could be?
If not, I can try to put together a minimal failing example, but I will need some time for that.

definelicht, 2022-01-26:
Hi there! This is a known issue, and seems to be a bug in the compiler. I opened an thread in the Xilinx support forums, but have so far been stonewalled:
https://support.xilinx.com/s/question/0D52E00006osTEASA2/internal-compiler-error-when-wrapping-hlsstreamhlsaxis-in-templated-class

From what I heard, the HLS team has seen this, and it might (?) be fixed in a future release, but unfortunately I don't have any promises or timetable.

I'm afraid you're gonna have to somehow work around it for now :-/

derpda, 2022-01-26:
Ok, thanks for letting me know so quickly!

The workaround is simple, I just use a `#if !defined(HLSLIB_SYNTHESIS)` statement to switch between the two stream classes (I want `hlslib::Stream` for simulation), and use the `read` and `write` member functions for access.

closed 2022-01-26 by derpda