definelicht/hlslib #60: No platform found in path

charliechou1001, 2022-03-10:
Hi, 
    When I run the gemm_hls project, the system reported an error:"No platform found in path /home/***/Xilinx/Vitis/2021.1/platforms", how can I solve the problem? Thank you!

cmake ../ -DMM_DATA_TYPE=float -DMM_PARALLELISM_N=32 -DMM_PARALLELISM_M=8 -DMM_MEMORY_TILE_SIZE_N=512 -DMM_MEMORY_TILE_SIZE_M=512

-- Using user defined Xilinx Runtime (XRT) directory "/home/***/Xilinx/XRT-master/build/Debug/opt/xilinx/xrt".
-- Looking for CL_VERSION_2_2
-- Looking for CL_VERSION_2_2 - found
-- Found OpenCL: /usr/lib/x86_64-linux-gnu/libOpenCL.so (found version "2.2") 
-- Found Vitis: /home/***/Xilinx/Vitis/2021.1/bin/v++  
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Looking for sgemm_
-- Looking for sgemm_ - found
-- A library with BLAS API found.
-- Querying default clock for xilinx_u250_gen3x16_xdma_3_1_202020_1.
No platform found in path /home/***/Xilinx/Vitis/2021.1/platforms
CMake Error at CMakeLists.txt:124 (string):
  string begin index: -1 is out of range 0 - 0

-- Querying Vitis platform for MatrixMultiplication.
No platform found in path /home/***/Xilinx/Vitis/2021.1/platforms
CMake Warning at hlslib/cmake/FindVitis.cmake:438 (message):
  Xilinx platform xilinx_u250_gen3x16_xdma_3_1_202020_1 was not found.
  Please consult "/home/***/Xilinx/Vitis/2021.1/bin/platforminfo -l" for a
  list of installed platforms.
Call Stack (most recent call first):
  CMakeLists.txt:198 (add_vitis_program)

definelicht, 2022-03-10:
Hi there. Which FPGA do you want to build for?

charliechou1001, 2022-03-11:
I'm running the script on the server, which has a U250 card, as well as other types of cards. I also tried to type "platforminfo -l" instruction, but the output shows as below:

{
    "platforms": ""
}

It seems that you use the code below to search for the available FPGA, is there any way to fix the problem? Thank you!

"find_program(Vitis_PLATFORMINFO platforminfo PATHS ${VITIS_ROOT}/bin NO_DEFAULT_PATH)"

definelicht, 2022-03-11:
It seems like this is not an hlslib issue. If `platforminfo` cannot find the platform then the tools also will not be able to. I recommend that you check if the U250 platform drivers have been installed, and then check where they are installed. Maybe it will help you find out why `platforminfo` does not detect them.

charliechou1001, 2022-03-11:
Would you mind telling me a bit more about the platform driver? Is there any user guide about it? And if I want to run the code on a machine without FPGA connected, but platform drivers are installed, can I still synthesize the code and build the host code?

definelicht, 2022-03-11:
They are here: https://www.xilinx.com/products/boards-and-kits/alveo/u250.html#gettingStarted
Yeah, you can still build everything on a machine without the FPGA, but you still need to install the platform from that page. The only thing you don't install is the bitstream onto the card itself.

charliechou1001, 2022-03-14:
Thanks a lot. I downloaded U250 Development Target Platform and then it works well during "make hw" step, though the hardware synthesis, especially block-level synthesis for gemm_hls, takes a long time.

 But I found it report an error when I change the platform to U50, the error was reported as follows, have you met a similar problem?

#make hw
.......
[ 50%] Linking MatrixMultiplication for hardware.
........

INFO: [SYSTEM_LINK 82-51] Create system connectivity graph
INFO: [SYSTEM_LINK 82-102] Applying explicit connections to the system connectivity graph: /home/xxx/code/gemm_hls/build/_x/link/sys_link/cfgraph/cfgen_cfgraph.xml
INFO: [SYSTEM_LINK 82-38] [03:44:33] cfgen started: /opt/tools/xilinx/Vitis/2020.2/bin/cfgen  -sp MatrixMultiplicationKernel_1.m_axi_gmem0:DDR[1] -sp MatrixMultiplicationKernel_1.m_axi_gmem1:DDR[1] -sp MatrixMultiplicationKernel_1.m_axi_gmem2:DDR[1] -dmclkid 0 -r /home/xxx/code/gemm_hls/build/_x/link/sys_link/_sysl/.cdb/xd_ip_db.xml -o /home/xxx/code/gemm_hls/build/_x/link/sys_link/cfgraph/cfgen_cfgraph.xml
INFO: [CFGEN 83-0] Kernel Specs: 
INFO: [CFGEN 83-0]   kernel: MatrixMultiplicationKernel, num: 1  {MatrixMultiplicationKernel_1}
INFO: [CFGEN 83-0] Port Specs: 
INFO: [CFGEN 83-0]   kernel: MatrixMultiplicationKernel_1, k_port: m_axi_gmem0, sptag: DDR[1]
INFO: [CFGEN 83-0]   kernel: MatrixMultiplicationKernel_1, k_port: m_axi_gmem1, sptag: DDR[1]
INFO: [CFGEN 83-0]   kernel: MatrixMultiplicationKernel_1, k_port: m_axi_gmem2, sptag: DDR[1]
ERROR: [CFGEN 83-2287] --sp tag applied with an invalid sp tag: DDR[1]
ERROR: [CFGEN 83-2287] --sp tag applied with an invalid sp tag: DDR[1]
ERROR: [CFGEN 83-2287] --sp tag applied with an invalid sp tag: DDR[1]
ERROR: [CFGEN 83-2297] Please consult platforminfo <platform.xpfm path> for sptag information
ERROR: [CFGEN 83-2298] Exiting due to previous error
ERROR: [SYSTEM_LINK 82-36] [03:44:34] cfgen failed
........

definelicht, 2022-03-14:
I think the U50 only has HBM, so you'd have to change the memory mapping to use this instead. I don't have a "nice" interface to configure this, unfortunately, so for now you'd have to manually change this in CMakeLists.txt. You can open an issue with a feature request :-)

closed 2022-03-14 by definelicht

YingWu321, 2022-05-24:
Hi,
Recently, I installed Xilinx RunTime and Deployment Target Platform of U250 by following step1 to step3 in the website https://www.xilinx.com/products/boards-and-kits/alveo/u250.html#gettingStarted . When I run 
`platforminfo -l 
`
it shows an error
`    Error parsing platform file: /opt/xilinx/platforms/xilinx_u250_gen3x16_xdma_4_1_202210_1.xpfm
    Line number:16. Mandatory element missing
    ERROR: Caught an internal exception: failed to parse the XPFM file
`
I think it might be the reason that I failed to run the cmake command
`-- Querying default clock for xilinx_u250_gen3x16_xdma_4_1_202210_1.
Exception: failed to parse the XPFM file
CMake Error at CMakeLists.txt:129 (string):
  string begin index: -1 is out of range 0 - 174

-- Querying Vitis platform for MatrixMultiplication.
Exception: failed to parse the XPFM file
-- Configuring incomplete, errors occurred!
`

In fact, xilinx_u250_gen3x16_xdma_4_1_202210_1.xpfm is in path "/opt/xilinx/platforms/xilinx_u250_gen3x16_xdma_4_1_202210_1/xilinx_u250_gen3x16_xdma_4_1_202210_1.xpfm". However, Vitis could not find it even though I copied it to "/opt/xilinx/platforms" manually.

I am wondering if there are any necessary operations to make xpfm file visible to Vitis. And I use Vitis 2021.1 on Ubuntu20.04.

​