definelicht/hlslib #62: `hlslib::Stream::write` works like `hls::stream::write`

derpda, 2022-04-18:
Hey, hope you're doing well!

### Intro
I will be holding a (very brief) presentation at a [small FPGA conference](https://fpgax.connpass.com/event/152964/) in Japan (sorry, info all in Japanese).
As it is just 5 minutes, I figured I would talk about the simulation capabilities of HLSLIB, in particular the dataflow/stream part since I did some work on that with you and am using it extensively in my project.
It seems easy enough to explain, and was very helpful during debugging and is something I think people should be using over the default `hls::stream` behavior.

### Issue
In the presentation, I want to show issues with the defaul `hls::stream` class, namely 
 - that it deadlocks when it shouldn't due to it being single-threaded,
 - that it doesn't deadlock when it should due to streams being of infinite depth

Now, I was defining some very simple example code, stuff like
```c++
void pipe_in(const float in_data[N], stream_t& in_stream_1, stream_t& in_stream_2) {
  for (int i=0; i<N; ++i) write(in_data[i]);
  for (int i=0; i<N; ++i) write(in_data[i]);
}
```
where I define `stream_t` as either `hlslib::Stream` or `hls::stream`. As you can see, I am using the `::write` interface so that I can just write the function once, as opposed to the `::Push` interface (which I normally prefer).
Good thing `hlslib::Stream` also conforms to that! :+1: \
However, it seems to not only be supporting the interface, but also to simulate the behavior of `hls::stream`, namely infinite depth:
https://github.com/definelicht/hlslib/blob/a67e02f263d666e1c7ba649c6755025b5fdd38bf/include/hlslib/xilinx/Stream.h#L172-L176

This means that hlslib will not deadlock when it really should when using this interface - is that intended?\

### Suggestion
I think it may be reasonable to have `hlslib::Stream::Push` and `hlslib::Stream::write` behave exactly the same, and only keep the latter for the desired interface compatibility.

definelicht, 2022-04-19:
Hey there! This is intended: My rationale with the compatibility interface is that it changes nothing from if you had been using `hls::stream`, such that users would experience no difference in behavior when changing their data type. Rather, the difference comes when using the hlslib API.

I see the argument for either approach, so I'm on the fence. If it's just for the sake of the presentation, I would lean towards not changing the behavior, because it's not _really_ the intended API anyway, so it doesn't seem worth the risk of breaking some existing code :-)

derpda, 2022-04-20:
You are right, we certainly shouldn't change it for the sake of a presentation! =D

That being said, I do believe we should change it so let me make my case.\
I think the purpose of the dataflow macros combined with `hlslib::Stream` is the realistic simulation of hardware to decrease development time by catching deadlocks. In particular, `hls::Stream` can be used as a drop-in replacement without changing the rest of the code. As it stands however, if a user does indeed just replace `hls::stream` with `hlslib::Stream` they do not in fact get the realistic behavior - they still get an infinite FIFO, which obviously can't be put into hardware.

Out (can I call it our? :D) readme also states that
>[`hls::Streams` are] bounded, simulating the finite capacity of hardware FIFOs, allowing easier detection of deadlocks in software

A reader would thus expect finite FIFOs, so I would argue that changing the `::write` method to behave like `::Push` would not break, but indeed *fix* existing code - paradoxically by potentially causing deadlocks.

As an additional note, I wasn't aware of this difference until I stumbled over the issue I laid out in my first post, and actually read the source code more in depth. The difference is just an additional parameter being passed to the same class method, so it is easy to miss.

definelicht, 2022-04-23:
Okay, sure! Feel free to push a PR :-)

derpda, 2022-05-13:
Closing this since its merged!

closed 2022-05-13 by derpda