definelicht/hlslib #64: Vitis 2021.1 failed to parse XPFM file of U250

YingWu321, 2022-05-24:
Hi,
I use gemm_hls project. Recently, I installed Xilinx RunTime and Deployment Target Platform of U250 by following step1 to step3 in the website https://www.xilinx.com/products/boards-and-kits/alveo/u250.html#gettingStarted . When I run 
> platforminfo -l 

it shows an error
> Error parsing platform file: /opt/xilinx/platforms/xilinx_u250_gen3x16_xdma_4_1_202210_1.xpfm
> Line number:16. Mandatory element missing
> ERROR: Caught an internal exception: failed to parse the XPFM file

I think it might be the reason that I failed to run the cmake command
> -- Querying default clock for xilinx_u250_gen3x16_xdma_4_1_202210_1.
> Exception: failed to parse the XPFM file
> CMake Error at CMakeLists.txt:129 (string):
>   string begin index: -1 is out of range 0 - 174
> 
> -- Querying Vitis platform for MatrixMultiplication.
> Exception: failed to parse the XPFM file
> -- Configuring incomplete, errors occurred!

In fact, xilinx_u250_gen3x16_xdma_4_1_202210_1.xpfm is in path "/opt/xilinx/platforms/xilinx_u250_gen3x16_xdma_4_1_202210_1/xilinx_u250_gen3x16_xdma_4_1_202210_1.xpfm". However, Vitis could not find it even though I copied it to "/opt/xilinx/platforms" manually.

I am wondering if there are any necessary operations to make xpfm file visible to Vitis. And I use Vitis 2021.1 on Ubuntu20.04.

​

definelicht, 2022-05-24:
Hi there :-) This issue is not related to hlslib, so I would suggest asking on the Xilinx support forums

YingWu321, 2022-05-25:
Hi,
The Xilinx support forum told me that
> The U250 XDMA 4_1 platform is not supported in Vitis 2021.1, please upgrade to Vitis 2022.1

I am wondering if the gemm_hls project supports Vitis 2022.1 and xilinx_u250_gen3x16_xdma_4_1_202210_1. I know I should change MM_PLATFORM in CMakeLists.txt. What else should I change to fit the new version?

definelicht, 2022-05-26:
The GEMM code is currently broken with 2021.2 and 2022.1, unfortunately. hlslib should still work, however.

I hope I will find time to look into what is causing the issue with the GEMM code, but I cannot promise anything at the moment.

closed 2022-06-29 by definelicht