definelicht/hlslib #70: i want to use your matrixmultiplication kernel !

CodeKwon, 2025-01-02:
I am working on a project that uses multiple kernels, including a matmul kernel. However, the matmul kernel I created doesn't perform very well, so I would like to use the matmul kernel you developed. How can I use it?

If I run make hw, can I obtain your kernel's .xo file? Would I be able to use that?

For reference, I am working on my project using the Vitis IDE GUI.

definelicht, 2025-01-12:
Sure, that should be possible. You mean from [this repository](https://github.com/spcl/gemm_hls), right?

Keep in mind that this matrix multiplication kernel is meant for very large matrices, so it will not perform well on small matrices 🙂

dfadfajl, 2025-01-17:
thank you for replying!

I'm planning to use the Scalable Matrix Multiplication project, but I'm not sure how to use it.

The kernels I created consist of a single .cpp file, and I connect these kernels in host.cpp.

However, the Scalable Matrix Multiplication project has a large number of files just to construct a single kernel, so I'm not sure how to use it in the Vitis GUI.

Could you guide me on how to set it up?

definelicht, 2025-01-20:
You would need to build it as a separate kernel `.xo` file, and then connect it to the right AXI ports during the linking stage. You can find documentation on how to do this in the Vitis documentation 🙂