definelicht/hlslib #72: Issues with automated part number extraction logic

salehjg, 2025-03-20:
Hi there,

After a year or two I have decided to give AWS F1 another try.
It seems that the logic to extract the FPGA partnumber from emconfigutil-generated json file fails with:

```
ERROR: JSON Key not found for filter "hardwarePlatform.board.part"
```

For the record, here is the content of the said json file:
```
{
    "Comment": "This file is auto-generated by the tool. Do not modify",
    "Version": {
        "FileVersion": "2.0",
        "ToolVersion": "2024.1"
    },
    "Platform": {
        "Boards": [
            {
                "Devices": [
                    {
                        "Name": "xilinx_aws-vu9p-f1_shell-v04261818_201920_4",
                        "DdrBanks": [
                            {
                                "Name": "mem0",
                                "Type": "ddr4",
                                "Size": "16GB"
                            },
                            {
                                "Name": "mem1",
                                "Type": "ddr4",
                                "Size": "16GB"
                            },
                            {
                                "Name": "mem2",
                                "Type": "ddr4",
                                "Size": "16GB"
                            },
                            {
                                "Name": "mem3",
                                "Type": "ddr4",
                                "Size": "16GB"
                            }
                        ]
                    }
                ],
                "NumBoards": "1"
            }
        ],
        "UnifiedPlatform": "true",
        "ExpandedPR": "false"
    }
}

```

Furthermore, here is my minimal CMakeLists.txt:
```
cmake_minimum_required(VERSION 3.20)
project(HelloWorld)

message("${CMAKE_MODULE_PATH}")
set(CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH} ${CMAKE_SOURCE_DIR}/hlslib/cmake)
message("${CMAKE_MODULE_PATH}")
find_package(Vitis REQUIRED)

add_executable(MyHostExecutable main.cpp)
include_directories(${Vitis_INCLUDE_DIRS})
target_link_libraries(MyHostExecutable ${Vitis_LIBRARIES})

# Will populate the "hw", "hw_emu", and "synthesis" targets
add_vitis_kernel(MyKernel FILES kernel.cpp)
add_vitis_program(MyKernel "xilinx_aws-vu9p-f1_shell-v04261818_201920_4")
```

I use FPGA (Ubuntu) AMI on AWS EC2 which has Vitis 2024.1.
Is this a known issue? Any advice?

salehjg, 2025-03-20:
So, its seems that this line is the culprit:
https://github.com/definelicht/hlslib/blob/f45eb33966999ce9c84a9964870d828b3b569d73/cmake/FindVitis.cmake#L479

Even when the part number is set manually with `CACHE INTERNAL` variable, the conditional is evaluated as True.
```
set(PROGRAM_PLATFORM_PART "xcvu9p-flga2104-2-e" CACHE INTERNAL "F1 part number")
...
set(CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH} ${CMAKE_SOURCE_DIR}/hlslib/cmake)
find_package(Vitis REQUIRED)
add_vitis_kernel(MyKernel FILES kernel.cpp)
add_vitis_program(MyKernel "xilinx_aws-vu9p-f1_shell-v04261818_201920_4")
```

salehjg, 2025-03-20:
Ok, it seems that changing the conditional to this, fixes the issue:
```
  if((NOT DEFINED PROGRAM_PLATFORM_PART) AND (NOT PROGRAM_PLATFORM_PART OR NOT "${${PROGRAM_TARGET}_PLATFORM}" STREQUAL "${PROGRAM_PLATFORM}"))
```
This is just to force it to be False if `PROGRAM_PLATFORM_PART` is defined before.

definelicht, 2025-03-20:
Please open a PR and I'll be happy to review 🙌